SasaKaranovic/winfidel-sensor #2: LittleFS mount fail on new device

cad435, 2024-05-12:
Hello!

After building the devices (PnP as per BOM, used JLCPCBA) i get the following error:

```Starting...
*wm:AutoConnect
*wm:Connecting to SAVED AP: ***
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.178.72
Connected.
Starting LittleFS...E (11254) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1367:error: Corrupted dir pair at {0x0, 0x1}

E (11255) esp_littlefs: mount failed,  (-84)
E (11258) esp_littlefs: Failed to initialize LittleFS
[ 11148][E][LittleFS.cpp:98] begin(): Mounting LittleFS failed! Error: -1
An Error has occurred while mounting LittleFS```

Any thoughts what went wrong?

cad435, 2024-05-12:
Just realized I forgot to build and upload the filesystem contents...

PEBCAK error I suppose :)

closed 2024-05-12 by cad435

SasaKaranovic, 2024-05-12:
Been there done that. :)
Glad you figured it out. Let me know how it works.

Would you mind sharing how much did the PCBA run cost?
I know there are others that are interested in ordering these. I received couple of emails asking how much would it cost to make 1/5/10 of these through one of the online PCB services.

cad435, 2024-05-12:
Sure!

5 pcs cost me 36.84€ shipped.

Mind you, the USB-C connector, ADC, LDO, the tactile swithces and the 8pf capacitors I soldered myself. I have those laying around and did not want to spend the additional cost on those.
The problem is, those are "non standard" components for JLC so they take 3$ setup fee per component, which would increase the cost by 15$. 
I was willing to take that for the ESP as I had not laying those around.

if you go all in I'd suppose you'll end up on about 50-60€ shipped for 5 pcs and about 90-100€ for 10pcs.

SasaKaranovic, 2024-05-12:
Awesome! Thank you for sharing this. I'm sure others will be happy to have a rough estimate on how much everything cost.