SasaKaranovic/winfidel-sensor #3: Hall Sensor Reversed

cad435, 2024-05-12:
It should be noted somewhere, that the hall sensor should be mounted the other way round, compared to the oroginal INFIDEL Sensor.

Top-PCB is the original, bottom PCB is WinFIDEL

[image]

Therefore the Hall-effect sensor must be glued in "upside down" compared to the original.

 - cad435

SasaKaranovic, 2024-05-13:
Hey @cad435 

Please note that while WInFiDEL is inspired by InFiDEL, it is not identical, obviously.
If you flip the WInFiDEL board, you should see the 3V3, GND and OUT markings for that IC.

There are two ways you can figure out which components is on which side.
- Manually: If you use silkscreen as your guide, there is silkscreen on top AND bottom side. On the bottom side there is silkscreen that indicates the sensor pinout and also U3 that indicates that hall effect sensor is on the bottom.
- Production: Pick and Place file indicates that the U3 is on the bottom side.

[image]

SasaKaranovic, 2024-05-13:
I don't have the InFiDEL sensor but just looked at some pictures and it looks like the sensor is mounted in the exact same way on the exact same side.

Can you clarify what do you mean by `the hall sensor should be mounted the other way round` ?
Are you using the Hall effect sensor that is specified by WInFiDEL design (ie `DRV5055A3`/`DRV5055A2`/`DRV5055A1`) ?

cad435, 2024-05-13:
Yes, I use a DRV5505.

So basically I hat some IFIDELS laying around and "upgraded" them with WinFIDEL.

The problem is as followed:
While INFIDEL hast the VIN pin of the hall-Sensor on the right pin (in my top picture), WinFIDEL has the VIN pin on the Left side.

So in order to upgrade from INFIDEL to WinFIDEL I had to remove the HALL sensor and glue a new one in, to match the Pinout.

And it works now 👌🏼

SasaKaranovic, 2024-05-14:
Glad to hear that you got it working.

I'm not sure where did this suggestion that WInFiDEL and InFiDEL are pin-to-pin compatible came from?
These are two different designs that share some design features but that's pretty much it.

closed 2024-05-14 by SasaKaranovic