SasaKaranovic/winfidel-sensor #5: missing parts info

johndeere6110, 2024-06-17:
[image]

hello 

When I want to buy the PCB, I am asked what should be placed in these places and I cannot find what should be placed there, can you help me?

SasaKaranovic, 2024-06-18:
Please take a look at the schematic and/or BOM.
Components that you marked are A1, C7, R11, C8 from left to right respectively.

johndeere6110, 2024-06-18:
I understand, but I can find it in the schematic, but it is not in the BOM so I don't know which capacitor should be there.
[Bill of Materials-ESP_Infidel.xlsx](https://github.com/user-attachments/files/15890868/Bill.of.Materials-ESP_Infidel.xlsx)
[image]

SasaKaranovic, 2024-06-20:
@johndeere6110 those parts are marked as DNP or Do Not Populate. Also sometimes interpreted as do not purchase/place etc) and they will not appear in the BOM.
So you would not solder these components on the board.

You might ask "Why then put them in the design if you know you are not going to populate them?"
There are many valid reasons to do so. In this particular case these are placeholders to be used used for antenna tuning.

closed 2025-01-11 by SasaKaranovic