SasaKaranovic/winfidel-sensor #7: Integration in Klipper

CBoismenu, 2025-02-15:
Recently bought one of your PCB's and constructed the "Winfidel".

It works perfectly.

Would like to integrate it with klipper via USB but can't seem to find information on how to do that.

Would you happen to have any kind of information on the subject.

Thanks in advance for your help.

SasaKaranovic, 2025-02-22:
Hey @CBoismenu 

I have just added a documentation for the [web API](https://docs.sasakaranovic.com/winfidel/web_api/).

Couple of people have also asked about using the WInFiDEL through USB/Serial instead of web API so I have updated the firmware to also print out the diameter over USB/serial ([docs](https://docs.sasakaranovic.com/winfidel/usb_api/)].

Let me know if you need any help or support from my side.
Also if you could share you Klipper integration with others, that would be greatly appreciated.

CBoismenu, 2025-02-22:
Documentation is interesting.

I will have to investigate furthermore if and how I would implement that in klipper.

I must admit though that my knowledge on how to do that is somewhat lacking.

I I find anything usefull I will relay the information.

Again thanks for your help.