SasaKaranovic/winfidel-sensor #8: ERROR while compileing

LoPh02, 2025-02-27:
if i try to compile i get allways the same error message:

>  PS C:\Users\Loren\winfidel-sensor-master\Firmware\platformio> pio run --target upload --upload-port COM3
> Processing esp32c3 (framework: arduino; platform: https://github.com/platformio/platform-espressif32.git; board: esp32-c3-devkitm-1)
> ------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> 
> *** missing SConscript file 'minify_data.py'
> File "C:\Users\Loren\AppData\Local\Programs\Python\Python313\Lib\site-packages\platformio\builder\main.py", line 167, in <module>

SasaKaranovic, 2025-03-02:
Try running `pio run` from `Firmware` directory, not platforimo.
Also make sure you have checked out entire project and also installed all the Python dependencies.

LoPh02, 2025-03-02:
thanks for the reply the hint that I forgot a Python dependenci has brought me further

closed 2025-03-02 by LoPh02

SasaKaranovic, 2025-03-02:
Glad you got it working!